autosportlabs/RaceCapture-Pro_firmware #1108: guard against divide by 0 if lat/lon does not change during interpolation

brentpicasso, 2021-12-22:
When interpolating lat/lon values, if the values do not change there is a risk of a divide by zero error:

Pseudo code
for i in range (0, interval_count):
            # Linearly interpolate longitude from latitude
            if (lat1 == lat2) and (lon1 == lon2):
                # skip past points that aren't changing
                continue
            interp_lon = lon1 + (interp_lat - lat1) * ((lon2 - lon1) / (lat2 - lat1))

RodADavison, 2021-12-22:
You only need to lest lat2 == lat1 to guard against x/0.  I spent a couple of minutes trying to figure out what this code does, but I'm still not sure!?  But your guard "protects" against any movement along a line of lat or lon, not just "points that aren't changing".

brentpicasso, 2021-12-22:
You are correct, I entered this issue early as a placeholder reminder to fix the firmware. 

 I was implementing the app-based lap recalculation, and as I was debugging it I saw I only needed to guard against non-changing latitude values.

RodADavison, 2021-12-22:
But code that disallows interopolation along lines of lat seems flawed.

brentpicasso, 2022-01-11:
Resolved by https://github.com/autosportlabs/RaceCapture-Pro_firmware/pull/1109

closed 2022-01-11 by brentpicasso